krakenrf/krakensdr_doa #3: EEPROM script minor suggestions

devnulling, 2021-12-24:
Example output from running the eeprom flashing util:

In the first line, `conectent` is misspelled, not sure what it's suppose to be? 

Additionally, the script as it calls out channels numbers are indexed as 0. (0-3)

```Please turn off all channels excepting channel number:0```

The dip switches are labeled 1-4, and the serial numbers are 1000-1003.

Might be better to have the script call out the DIP switch labels (1-4) that should be on, and/or a serial number scheme of 1001-1004 ? 

```
Initialzing EEPROM conectent and setting serial numbers
Found 4 receivers
Do you wish to overwrite the current EEPROM content?y
yes
Please turn off all channels excepting channel number:0 and press enter
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
Realtek default OEM with EEPROM
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		1000
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Please turn off all channels excepting channel number:1 and press enter
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		002
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
Realtek default OEM with EEPROM
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		1001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Please turn off all channels excepting channel number:2 and press enter
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		003
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
Realtek default OEM with EEPROM
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		1002
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Please turn off all channels excepting channel number:3 and press enter
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		004
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
Realtek default OEM with EEPROM
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
Found 1 device(s):
  0:  Generic RTL2832U OEM

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T/2 tuner

Current configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		Realtek
Product:		RTL2838UHIDIR
Serial number:		00000001
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________

New configuration:
__________________________________________
Vendor ID:		0x0bda
Product ID:		0x2838
Manufacturer:		RTL-SDR
Product:		KerberosSDR
Serial number:		1003
Serial number enabled:	yes
IR endpoint enabled:	yes
Remote wakeup enabled:	no
__________________________________________
Write new configuration to device [y/n]? y

Configuration successfully written.
Please replug the device for changes to take effect.
EEPROM writing script finished. Plese perform a full power cycle.
```

aaknitt, 2022-01-01:
In addition to the above, I've noticed that the channel numbers in the legend of the Spectrum GUI screen (http://krakensdr:8050/spectrum) don't seem to match the EEPROM serial numbers.  On my unit, I'm seeing the following:

Serial number 1000 shows as Channel 2 on the spectrum plot
Serial number 1001 shows as Channel 3 on the spectrum plot
Serial number 1002 shows as Channel 0 on the spectrum plot
Serial number 1003 shows as Channel 1 on the spectrum plot

This begs the question as to which device/port/channel should map to each antenna in the array as shown here:
https://www.rtl-sdr.com/wp-content/uploads/2019/05/kerberossdr_UCA_diagram.png?ffccfa&ffccfa

Is the Spectrum plot an accurate depiction of which channel should correspond to antenna position, or should the EEPROM serial number be used to determine antenna position?

I'm getting very poor initial results with DOA estimation using a UCA and I'm wondering if it may be because I have the antennas incorrectly ordered?  The results often seem off by 45 or 180 degrees with sudden jumps as I change quadrants, so I'm wondering if it may be an antenna sequence error?  

[image]

Infiltra, 2022-01-05:
@krakenrf I also faced the same issues which I rejected for some reason, I'd like to take this up and investigate further.

krakenrf, 2022-01-05:
@aaknitt this has been fixed in a recent update, but not on the premade image yet. The only thing it affects is the spectrum display, not the actual DOA results.

As you're running a Kerberos, make sure that you have the antennas disconnected when calibrating, and if you're using a close range transmitter for testing, make sure that is OFF when calibrating as leakage into the calibration hardware network could also cause skewing issues. 

Also remember that at your test freq of 146 MHz, you're array size will necessarily be quite large. A 35cm radius is probably the smallest you can do. If you're testing indoors, this freq will probably results in a lot of reflections from walls too, and your TX should be at least a wavelength away. If you need further help, please post a new issue with full details and photos of your setup.

krakenrf, 2022-01-06:
Fixed in a recent push to heimdall_daq_fw

closed 2022-01-06 by krakenrf