krakenrf/krakensdr_doa #6: Use JSON for config files

devnulling, 2021-12-25:
There seems to be a mix of usage of JSON and INI file formats for config files.

https://github.com/krakenrf/krakensdr_doa/blob/clientside_graphs/settings.json

https://github.com/krakenrf/heimdall_daq_fw/blob/development/config_files/cw_15kHz/daq_chain_config.ini

Ideally should just use a single format and stick with it. JSON is superior IMO to INI file formats.

krakenrf, 2022-01-05:
Thanks. Two different devs contributed those config files which could explain the difference. We will be looking at converting to full JSON in a future update.

krakenrf, 2022-08-26:
Because modifying the ini is less important now that we moved most functionality to the DSP side, we're just going to leave it as is for now.