krakenrf/krakensdr_doa #11: Add option for only frequency of interest

devnulling, 2022-03-13:
The current behavior appears to be that the strongest signal within a pass band is selected for DOA. 

If there is only a single frequency of interest being DF'ed, this can lead to that frequency being ignored and/or a bunch of additional unwanted data.

It's mentioned here [0] [1] that multiple channels with independent squelches, which is a nice feature. It'd would be desirable to have a option/mode to ignore everything other than the frequency of interest though.

[0] - https://github.com/krakenrf/krakensdr_doa/tree/clientside_graphs#upcoming-features-and-known-bugs
[1] - 
```
[FEATURE] Currently squelch works by selecting the strongest signal that is active and above the set threshold within the active bandwidth. The next steps will be to allow users to create multiple channels within the active bandwidth, each with their own squelch. This will allow users to track multiple signals at once, and ignore unwated signals within the bandwidth at the same time.
```

krakenrf, 2022-03-29:
Yep the current version was designed with the idea that DAQ side decimation would be applied to reduce the bandwidth down to the signal of interest only. 

Next version we're working on will have the ability to run wideband, and create narrowband channel VFOs in the active spectrum, and also click to tune. So for the basic case of a single frequency, you can set the channel bandwidth, squelch value and then just click on the signal to tune it (or enter the freq manually)

There's a bit of a holdup with an issue on the DAQ side regarding off-center tuning. It is something to do with fractional sample corrections. But we've found an elegant fix which is in the process of being implemented by our DSP engineer.

krakenrf, 2022-03-29:
[image]

krakenrf, 2022-04-08:
Implemented in the main branch now.

closed 2022-04-08 by krakenrf