krakenrf/krakensdr_doa #17: Webpage (and app) Compass is backwards

DueHack12, 2022-07-03:
Hello,

I am trying to use the compass function from the webpage, and it is showing the radio in front of my antennas, even though the radio is behind it. Whenever we move it side to side, it sticks the the right.

krakenrf, 2022-07-11:
Is it possible to see a video?

Can you double check your antenna orientation matches what's shown here? https://github.com/krakenrf/krakensdr_docs/wiki/4.-Antenna-Array-Setup

Also are you testing indoors, and what frequency and array spacing are you using? If you test indoors or in a place with a lot of walls you can get weird results due to multipath.

krakenrf, 2022-08-01:
Closing as no reply, and not sure what the issue was.

closed 2022-08-01 by krakenrf

DueHack12, 2022-08-03:
Sorry, a re-flash of the firmware fixed it.