krakenrf/krakensdr_doa #18: df-aggregator does not work with the current version  V1.1 (270622)

IIvanIvanovich, 2022-07-03:
Hi! I tried running the RDF Data Simulator emulator and got a great job.
XML output looks like this :
NOCALL1656870732764120.039.128403-76.51774949.1671.20595
NOCALL1656872048861120.039.128844-76.51709249.112442.2186178
NOCALL1656872061033120.039.130298-76.51542727.910342.7049182
NOCALL1656872071124120.039.13193-76.5143825.83341.1826
NOCALL1656872081815120.039.13258-76.516287.933342.0049176

Station ID: NOCALL
Location: 39.129781°, -76.515815°
Heading: 35.8°
Tuned to 120 MHz

if I start the kraken_doa, I get it:
NOCALL1656871730414120.039.129781-76.51581535.8349-74.10.75
NOCALL1656871836612120.039.129781-76.51581535.8349-74.00.74
NOCALL1656871858866120.039.129781-76.51581535.8350-73.90.92
NOCALL1656871867596120.039.129781-76.51581535.8195-74.00.72

You can clearly see that the formatting at the end of the line is different.
DF-aggregator does not work with this formatting.

I tried 190522 and 150422. They also do not work in this mode.
I saw that everything works in https://www.youtube.com/watch?v=KE2FRv-hoQs on the January 11, 2022 version.

In addition, you cannot change NOCALL To anything else

Awaiting your reply

IIvanIvanovich, 2022-07-04:
An error occurs in the df-aggregator when working with the CONFIDENCE variable, in debag I see:
invalid literal for int() with base 10: '2.05'
kraken_doa issues it as a float, and the df-aggregator expects her Int.

IIvanIvanovich, 2022-07-04:
in the old version, the calculation was as follows:

  confidence_str = "{}".format(np.max(int(float(confidence_str)*100)))
  max_power_level_str = "{:.1f}".format((np.maximum(-100, float(max_power_level_str)+100)))

In new is:

   confidence_str = "{:.2f}".format(np.max(conf_val))
   max_power_level_str = "{:.1f}".format((np.maximum(-100, max_amplitude)))

Explicit Code error, as well as the fact that you cannot change the callsign ID

ckoval7, 2022-07-05:
In the long term, I'm going to update DFA to work with the new Kraken formats. In the near term, I'm going to submit a PR to spit out values more in line with the Kerberos format.

ckoval7, 2022-07-05:
As for changing "NOCALL", look for "Station ID" in the KrakenSDR web interface.

krakenrf, 2022-07-11:
Should be fixed with https://github.com/krakenrf/krakensdr_doa/pull/22

Will look later at perhaps combining the Kerberos and DFA drop down, but keeping it separate might help avoid confusion.

closed 2022-07-11 by krakenrf