krakenrf/krakensdr_doa #24: Confirmation of dash version...

darksidelemm, 2022-07-11:
I'm attempting to set up krakensdr_doa on an intel NUC, due to having experienced big RFI issues with RPi4's.

After following the instructions [here](https://github.com/krakenrf/krakensdr_doa#manual-installation-from-a-fresh-os) verbatim, I end up at the point where i startup the software, but the main webserver on port 8080 doesn't start, with the following error:

```
(kraken) darkside@darkside-nuc:~/krakensdr/krakensdr_doa$ cat ui.log 
Traceback (most recent call last):
  File "/home/darkside/krakensdr/krakensdr_doa/_UI/_web_interface/kraken_web_interface.py", line 82, in <module>
    import tooltips
  File "/home/darkside/krakensdr/krakensdr_doa/_UI/_web_interface/tooltips.py", line 2, in <module>
    import dash_bootstrap_components as dbc
  File "/home/darkside/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_bootstrap_components/__init__.py", line 7, in <module>
    from dash_bootstrap_components._table import _generate_table_from_df
  File "/home/darkside/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_bootstrap_components/_table.py", line 5, in <module>
    from dash import html
ImportError: cannot import name 'html' from 'dash' (/home/darkside/miniforge3/envs/kraken/lib/python3.9/site-packages/dash/__init__.py)
```

I can confirm that the dash version is the one specified in the installation instructions (1.20.0):
```
>>> import dash
>>> dash.__version__
'1.20.0'
```

Is this actually the version that is required, or are these instructions out of date?

krakenrf, 2022-07-11:
Can you paste your "conda list" here? Seems to be something different with dash_bootstrap_components. Maybe they changed dependencies which breaks things.

On my working Kraken systems dash_bootstrap_components is version 1.1.0, try installing that version if you have 1.2.0.

darksidelemm, 2022-07-11:
Yep.. v1.2.0 for dash_bootstrap_components, which is what pip3 brought in.
Pinned that to 1.1.0 and i can now import html now, and the webserver starts.

However, now when I navigate to the web interface, I just get a black page, and the following error in the JS console:
[image]

```
(kraken) darkside@darkside-nuc:~/krakensdr/krakensdr_doa$ conda list
# packages in environment at /home/darkside/miniforge3/envs/kraken:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
aiofiles                  0.8.0              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.6.1              h7f98852_0    conda-forge
asttokens                 2.0.5              pyhd8ed1ab_0    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blas                      2.106                       mkl    conda-forge
blinker                   1.4                        py_1    conda-forge
bokeh                     2.4.3            py39hf3d152e_0    conda-forge
brotli                    1.0.9                    pypi_0    pypi
brotli-bin                1.0.9                h166bdaf_7    conda-forge
brotlipy                  0.7.0           py39hb9d737c_1004    conda-forge
ca-certificates           2022.6.15            ha878542_0    conda-forge
certifi                   2022.6.15        py39hf3d152e_0    conda-forge
cffi                      1.15.1           py39he91dace_0    conda-forge
charset-normalizer        2.1.0              pyhd8ed1ab_0    conda-forge
click                     8.1.3            py39hf3d152e_0    conda-forge
configparser              5.2.0              pyhd8ed1ab_0    conda-forge
cryptography              37.0.4           py39hd97740a_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dash                      1.20.0                   pypi_0    pypi
dash-bootstrap-components 1.2.0                    pypi_0    pypi
dash-core-components      1.16.0                   pypi_0    pypi
dash-devices              0.1.3                    pypi_0    pypi
dash-html-components      1.1.3                    pypi_0    pypi
dash-renderer             1.9.1                    pypi_0    pypi
dash-table                4.11.3                   pypi_0    pypi
dbus                      1.13.6               h5008d03_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
executing                 0.8.3              pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
flask                     2.1.2                    pypi_0    pypi
flask-compress            1.12                     pypi_0    pypi
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               h8e229c2_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.34.4           py39hb9d737c_0    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
future                    0.18.2           py39hf3d152e_5    conda-forge
gettext                   0.19.8.1          h73d1719_1008    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
glib                      2.70.2               h780b84a_4    conda-forge
glib-tools                2.70.2               h780b84a_4    conda-forge
gst-plugins-base          1.20.3               hf6a322e_0    conda-forge
gstreamer                 1.20.3               hd4edc92_0    conda-forge
h11                       0.13.0             pyhd8ed1ab_1    conda-forge
h2                        4.1.0            py39hf3d152e_0    conda-forge
hpack                     4.0.0              pyh9f0ad1d_0    conda-forge
hypercorn                 0.13.2           py39hf3d152e_0    conda-forge
hyperframe                6.0.1              pyhd8ed1ab_0    conda-forge
icc_rt                    2020.2                intel_254    numba
[... 123 lines cut ...]
tk                        8.6.12               h27826a3_0    conda-forge
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tornado                   6.2              py39hb9d737c_0    conda-forge
traitlets                 5.3.0              pyhd8ed1ab_0    conda-forge
traits                    6.3.2            py39hb9d737c_1    conda-forge
traitsui                  7.4.0              pyhd8ed1ab_0    conda-forge
trio                      0.21.0           py39hf3d152e_0    conda-forge
typing_extensions         4.3.0              pyha770c72_0    conda-forge
tzdata                    2022a                h191b570_0    conda-forge
unicodedata2              14.0.0           py39hb9d737c_1    conda-forge
[... cut ...]
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xz                        5.2.5                h516909a_1    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h166bdaf_1    conda-forge
zstd                      1.5.2                h8a70e8d_2    conda-forge
```

darksidelemm, 2022-07-11:
Ahh... pinning dash_bootstrap_components to 1.1.0 resulted in dash itself getting updated by pip3:
```
Successfully installed dash-2.5.1 dash-core-components-2.0.0 dash-html-components-2.0.0 dash-table-5.0.0
```

After re-running `conda install dash==1.20.0` and `conda install werkzeug==2.0.2` (in which conda complains a bit) I end up with a web interface that seems to work. 

Might be worth updating the documentation, because i'm sure others are going to hit this problem...

krakenrf, 2022-07-11:
Ah was just about the suggest that.

Yeah dash versioning is going to be an annoying problem with them depreciating a lot of things that dash_devices works with, and dash_devices seems abandoned these days. I guess we just pin everything to versions that work.

Although, I just tested a fresh install with pip3 install dash_bootstrap_components and got v1.2.0 and I had no problems running. So something weird is happening with conda and pip3 interacting weirdly on some systems.

closed 2022-07-11 by krakenrf