krakenrf/krakensdr_doa #27: Can't get a webpage, Ubuntu 20 on ODROID N2+

xmutantson, 2022-08-25:
Hey there, I don't think this specific error has been addressed yet. I am expecting the arrival of my kraken soon so I was setting up the software in advance. This is to say I do not have the krakensdr attached yet. Setup seemingly went fine up until actually starting the kraken_doa_start.sh file. Here's the output, which looks fine to me:

`Shut down DAQ chain ..
[sudo] password for kameron:
Config file check bypassed [ WARNING ]
kernel.sched_rt_runtime_us = -1
Desig FIR filter with the following parameters:
Decimation ratio: 1
Bandwidth: 1.00
Tap size: 1
Window function: hann
FIR filter ready
Transfer funcfion is exported to :  _logs/Decimator_filter_transfer.html
Coefficients are exported to:  _data_control/fir_coeffs.txt
Starting DAQ Subsystem
Output data interface: Shared memory
      )  (
      (   ) )
       ) ( (
     _______)_
  .-'---------|
 (  |/\/\/\/\/|
  '-./\/\/\/\/|
    '_________'
     '-------'

Have a coffee watch radar
Starting KrakenSDR Direction Finder
Web Interface Running at 0.0.0.0:8080
PHP Data Out Server Running at 0.0.0.0:8081
`

Then, nothing on port 8080. Connection refused. Since I saw that this info was helpful for the other issue about no webpage, here's the conda list and ~/krakensdr/krakensdr_doa$ cat ui.log:
`# packages in environment at /home/kameron/miniforge3/envs/kraken:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
aiofiles                  0.8.0              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.3.2              h4e544f5_0    conda-forge
asttokens                 2.0.8              pyhd8ed1ab_0    conda-forge
async_generator           1.10                       py_0    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
blinker                   1.4                        py_1    conda-forge
bokeh                     2.4.3              pyhd8ed1ab_3    conda-forge
brotli                    1.0.9                h4e544f5_7    conda-forge
brotli-bin                1.0.9                h4e544f5_7    conda-forge
brotli-python             1.0.9            py39h3d8bfb9_7    conda-forge
brotlipy                  0.7.0           py39h0fd3b05_1004    conda-forge
ca-certificates           2022.6.15            h4fd8a4c_0    conda-forge
certifi                   2022.6.15        py39h4420490_0    conda-forge
cffi                      1.15.1           py39hb26bf21_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3            py39h4420490_0    conda-forge
configparser              5.3.0              pyhd8ed1ab_0    conda-forge
cryptography              37.0.4           py39h32042e6_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dash                      2.6.1                    pypi_0    pypi
dash-bootstrap-components 1.1.0                    pypi_0    pypi
dash-core-components      2.0.0                    pypi_0    pypi
dash-devices              0.1.3                    pypi_0    pypi
dash-html-components      2.0.0                    pypi_0    pypi
dash-renderer             1.9.1              pyhd8ed1ab_0    conda-forge
dash-table                5.0.0                    pypi_0    pypi
dataclasses               0.8                pyhc8e2a94_3    conda-forge
dbus                      1.13.6               h12b9eeb_3    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
executing                 0.10.0             pyhd8ed1ab_0    conda-forge
[... 172 lines cut ...]
`Traceback (most recent call last):
  File "/home/kameron/krakensdr/krakensdr_doa/_UI/_web_interface/kraken_web_interface.py", line 35, in <module>
    import dash_devices as dash
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/__init__.py", line 1, in <module>
    from .dash import Dash, no_update, g_cc, Services  # noqa: F401
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/dash.py", line 18, in <module>
    import quart
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/__init__.py", line 5, in <module>
    from .app import Quart as Quart
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/app.py", line 49, in <module>
    from .asgi import ASGIHTTPConnection, ASGILifespan, ASGIWebsocketConnection
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/asgi.py", line 29, in <module>
    from .debug import traceback_response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/debug.py", line 8, in <module>
    from .wrappers import Response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/__init__.py", line 3, in <module>
    from .base import BaseRequestWebsocket
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/base.py", line 9, in <module>
    from .. import json
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/json/__init__.py", line 7, in <module>
    from ..globals import current_app
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/globals.py", line 16, in <module>
    app_ctx: _AppCtxGlobals = LocalProxy(  # type: ignore[assignment]
TypeError: __init__() got an unexpected keyword argument 'unbound_message'
`

xmutantson, 2022-08-25:
Ok, so looking into this a bit further, it seems that quart is the showstopper. The dash version shown was corrected by doing pip uninstall dash to get rid of the 2.6.0 version. That didn't make any difference to the ui.log error. So I kept checking the difference between my conda.list file and the one referenced in #24 and I found the following differences (only checking the ones requested by kraken instructions:

Quart, 0.17.0 vs my 0.18.0
orjson, 3.7.7 vs my 3.7.12
matplotlib, 3.5.2 vs my 3.5.3
dash 1.20.0 vs 2.6.1, fixed by above

Since quart was all over the traceback, I did python3 to open python, then import quart:

```
>>> import quart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/__init__.py", line 5, in <module>
    from .app import Quart as Quart
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/app.py", line 49, in <module>
    from .asgi import ASGIHTTPConnection, ASGILifespan, ASGIWebsocketConnection
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/asgi.py", line 29, in <module>
    from .debug import traceback_response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/debug.py", line 8, in <module>
    from .wrappers import Response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/__init__.py", line 3, in <module>
    from .base import BaseRequestWebsocket
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/base.py", line 9, in <module>
    from .. import json
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/json/__init__.py", line 7, in <module>
    from ..globals import current_app
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/globals.py", line 16, in <module>
    app_ctx: _AppCtxGlobals = LocalProxy(  # type: ignore[assignment]
TypeError: __init__() got an unexpected keyword argument 'unbound_message'
>>> 
```

Aha! that looks familiar. In retrospect this is pretty obvious, dash is trying to import quart in the main ui.log output but it's failing right there. So I tried to install quart to be the same version as step 1 assuming that 0.18.0 is broken.

```
(kraken) kameron@highwayman:~$ conda install quart==0.17.0
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                              

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - quart==0.17.0 -> python[version='>=2.7|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.10,<3.11.0a0|>=3.4|>=3.6|>=3.8,<3.9.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.9,<3.10.0a0|>=3.9,<3.10.0a0|>=3.7,<3.8.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.6,<3.7.0a0|>=3.7|>=3.5',build='*_cpython|*_73_pypy|*_cpython|*_73_pypy']

Your python: python=3.9.7

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

3.9.7 was the specific python version requested. also in #24 this is the version of python that conda list shows. So it seems like an impossible configuration. I guess I need help fixing this conflict. It may not be a version issue with quart but I also tried searching for the error on the quart side and didnt find anything to do with quart.

xmutantson, 2022-08-26:
I didn't notice that quart was installed with pip instead of conda. When removing via pip and reinstalling with conda, this is the output:

```
(kraken) kameron@highwayman:~$ conda install quart
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/kameron/miniforge3/envs/kraken

  added / updated specs:
    - quart

The following NEW packages will be INSTALLED:

  hypercorn          conda-forge/noarch::hypercorn-0.5.2-py_0
  quart              conda-forge/noarch::quart-0.8.0-py_0

Proceed ([y]/n)? 

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Retrieving notices: ...working... done
(kraken) kameron@highwayman:~$ python
Python 3.9.7 | packaged by conda-forge | (default, Oct 10 2021, 15:08:54) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import quart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/__init__.py", line 1, in <module>
    from jinja2 import escape, Markup
ImportError: cannot import name 'escape' from 'jinja2' (/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/jinja2/__init__.py)
>>> 
```

Seems that this is a super old version of quart. I am going to tear the environment down and restart the setup. If that fixes it I'll come back and recommend this be closed.

xmutantson, 2022-08-26:
Looks like a fresh environment does install this version of quart anyway. I started by removing the environment, then doing the whole environment setup in the heimdall section, then continuing in kraken_doa.

```(kraken) kameron@highwayman:~$ conda install quart
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/kameron/miniforge3/envs/kraken

  added / updated specs:
    - quart

The following NEW packages will be INSTALLED:

  aiofiles           conda-forge/noarch::aiofiles-0.8.0-pyhd8ed1ab_0
  async_generator    conda-forge/noarch::async_generator-1.10-py_0
  attrs              conda-forge/noarch::attrs-22.1.0-pyh71513ae_1
  blinker            conda-forge/noarch::blinker-1.4-py_1
  click              conda-forge/linux-aarch64::click-8.1.3-py39h4420490_0
  h11                conda-forge/noarch::h11-0.13.0-pyhd8ed1ab_1
  h2                 conda-forge/linux-aarch64::h2-4.1.0-py39ha65689a_0
  hpack              conda-forge/noarch::hpack-4.0.0-pyh9f0ad1d_0
  hypercorn          conda-forge/noarch::hypercorn-0.5.2-py_0
  hyperframe         conda-forge/noarch::hyperframe-6.0.1-pyhd8ed1ab_0
  idna               conda-forge/noarch::idna-3.3-pyhd8ed1ab_0
  itsdangerous       conda-forge/noarch::itsdangerous-2.1.2-pyhd8ed1ab_0
  libuv              conda-forge/linux-aarch64::libuv-1.44.2-h4e544f5_0
  multidict          conda-forge/linux-aarch64::multidict-6.0.2-py39h0fd3b05_1
  outcome            conda-forge/noarch::outcome-1.2.0-pyhd8ed1ab_0
  pytoml             conda-forge/noarch::pytoml-0.1.21-py_0
  quart              conda-forge/noarch::quart-0.8.0-py_0
  sniffio            conda-forge/linux-aarch64::sniffio-1.2.0-py39h4420490_3
  sortedcontainers   conda-forge/noarch::sortedcontainers-2.4.0-pyhd8ed1ab_0
  trio               conda-forge/linux-aarch64::trio-0.21.0-py39h4420490_0
  uvloop             conda-forge/linux-aarch64::uvloop-0.16.0-py39h14843e3_1
  wsproto            conda-forge/noarch::wsproto-1.2.0-pyhd8ed1ab_0

Proceed ([y]/n)? 

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Retrieving notices: ...working... done
```

I'm going to keep going.

xmutantson, 2022-08-26:
Here's a complete chain from creation of the environment all the way to ui.log showing the error and me replicating it just by trying to import quart:

The 0.18.0 version of quart gets pulled in with quart_compress via pip.

(I collapsed it because it's a huge paste)
<details>
  <summary>Click me</summary>
  ### Output
  ```
kameron@highwayman:~$ conda create -n kraken python=3.9.7
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/kameron/miniforge3/envs/kraken

  added / updated specs:
    - python=3.9.7

The following NEW packages will be INSTALLED:

  _openmp_mutex      conda-forge/linux-aarch64::_openmp_mutex-4.5-2_gnu
  ca-certificates    conda-forge/linux-aarch64::ca-certificates-2022.6.15-h4fd8a4c_0
  ld_impl_linux-aar~ conda-forge/linux-aarch64::ld_impl_linux-aarch64-2.36.1-h02ad14f_2
  libffi             conda-forge/linux-aarch64::libffi-3.4.2-h3557bc0_5
  libgcc-ng          conda-forge/linux-aarch64::libgcc-ng-12.1.0-h3242a24_16
  libgomp            conda-forge/linux-aarch64::libgomp-12.1.0-h3242a24_16
  libnsl             conda-forge/linux-aarch64::libnsl-2.0.0-hf897c2e_0
  libsqlite          conda-forge/linux-aarch64::libsqlite-3.39.2-hf9034f9_1
  libzlib            conda-forge/linux-aarch64::libzlib-1.2.12-h4e544f5_2
  ncurses            conda-forge/linux-aarch64::ncurses-6.3-headf329_1
  openssl            conda-forge/linux-aarch64::openssl-3.0.5-h4e544f5_1
  pip                conda-forge/noarch::pip-22.2.2-pyhd8ed1ab_0
  python             conda-forge/linux-aarch64::python-3.9.7-h47f6e27_4_cpython
  python_abi         conda-forge/linux-aarch64::python_abi-3.9-2_cp39
  readline           conda-forge/linux-aarch64::readline-8.1.2-h38e3740_0
  setuptools         conda-forge/linux-aarch64::setuptools-65.3.0-py39ha65689a_0
  sqlite             conda-forge/linux-aarch64::sqlite-3.39.2-h69ca7e5_1
  tk                 conda-forge/linux-aarch64::tk-8.6.12-hd8af866_0
  tzdata             conda-forge/noarch::tzdata-2022c-h191b570_0
  wheel              conda-forge/noarch::wheel-0.37.1-pyhd8ed1ab_0
  xz                 conda-forge/linux-aarch64::xz-5.2.6-h9cdd2b7_0

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate kraken
#
# To deactivate an active environment, use
#
#     $ conda deactivate

Retrieving notices: ...working... done
kameron@highwayman:~$ conda activate kraken
(kraken) kameron@highwayman:~$ conda install scipy
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /home/kameron/miniforge3/envs/kraken

[... 592 lines cut ...]
>>> import quart
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/__init__.py", line 5, in <module>
    from .app import Quart as Quart
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/app.py", line 49, in <module>
    from .asgi import ASGIHTTPConnection, ASGILifespan, ASGIWebsocketConnection
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/asgi.py", line 29, in <module>
    from .debug import traceback_response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/debug.py", line 8, in <module>
    from .wrappers import Response
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/__init__.py", line 3, in <module>
    from .base import BaseRequestWebsocket
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/wrappers/base.py", line 9, in <module>
    from .. import json
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/json/__init__.py", line 7, in <module>
    from ..globals import current_app
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/globals.py", line 16, in <module>
    app_ctx: _AppCtxGlobals = LocalProxy(  # type: ignore[assignment]
TypeError: __init__() got an unexpected keyword argument 'unbound_message'
>>> 
(kraken) kameron@highwayman:~/krakensdr/krakensdr_doa$ 

  ```
</details>

xmutantson, 2022-08-26:
Ugh, after all that all I had to do was pip uninstall quart and then pip install quart==0.17.0

Resolved.

Current output of ui.log (not sure if what is breaking is important):

```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/kameron/miniforge3/envs/kraken/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/home/kameron/krakensdr/krakensdr_doa/_signal_processing/krakenSDR_signal_processor.py", line 233, in run
    single_ch = self.processed_signal[1, :]
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
```

krakenrf, 2022-08-26:
I have an Odroid N2 here, let me try to reinstall from scratch, I had it working on that device previously without issue. These packages do seem to often update without a care for backwards compatibility though. We might have to pin everything to working versions.

pip quart_compress shouldn't be pulling quart, it doesn't do that on any other system I've tried recently, so that's also really weird.

Will be back in a few hours with my N2 install results.

krakenrf, 2022-08-26:
Ok so I came across the same issue you have. I'm not sure why only this distro pulls the incompatible quart with quart_compress, but doing pip install quart==0.17.0 fixes it. I've updated the install docs to use this method now.

But there's an even bigger problem with the N2. For some reason the USB ports on the N2 just aren't stable with this many USB tuners. I've been trying to see if there is a solution, but I can't find one yet.

If you Google for Odroid N2 USB issues you'll find a lot of similar complaints. It must have been the XU4 I had it working on before, there was a reason I packed away the N2.

On the Odroid Ubuntu MATE 20.04 LTS distro without any updates, even rtl_test won't run correctly. And the USB devices are constantly showing up and then going away in lsusb too. After updating `sudo apt full-upgrade`, rtl_test and lsusb become stable. But when we try to start all the RTL-SDRs via heimdall the Odroid starts throwing USB errors.

xmutantson, 2022-08-26:
Is your board an N2 or N2+? it seems that they had some production issues but made some changes to the board layout. I've previously had 2 rtl-sdrs running in parallel for months on end and also 3 wifi cards, one of which was running in monitor mode. Thats a lot of USB I guess but so far I haven't had any dropouts. 

I think I'm running normal ubuntu, not ubuntu mate.

```kameron@highwayman:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.4 LTS
Release:        20.04
Codename:       focal
```

xmutantson, 2022-08-26:
Oops, forgot to attach hardkernel's breakdown of the issue:
https://wiki.odroid.com/odroid-n2/hardware#n2_pcb_revision_history

My board is an odroid n2+, but I'm unsure about the revision. I had an eMMC module installed and it has a thermal pad that rips the stricker off right where the production info was. Bad design on purpose maybe?

krakenrf, 2022-08-26:
I have an older N2 and a newer N2+. Using the newer N2+ at the moment. I got it working. I'm not sure exactly what helped, but right on my working image I'm running:

- Armbian Odroid Desktop Version
- Flashed the latest spiboot firmware

And maybe this was it all along: It only works if plugged into the USB ports closest to the DC power barrel jack. Using the other USB ports results in constant USB failure.

Will test raw Ubuntu soon too.

krakenrf, 2022-08-26:
Where did you download your Ubuntu image from?

I went back to Odroid's Ubuntu MATE release, and confirmed that the USB doesn't work properly in that OS, even when on the 'good' USB ports. I think it has something to do with it not recognizing that the RTL-SDRs connect via a USB hub in the Kraken.

Armbian seems to work but only when connected to the 'good' DC barrel side USB ports. I don't know what the long term stability of it is though. The USB ports just seem very buggy on the N2.

krakenrf, 2022-08-26:
Unfortunately it's not stable, within 5 minutes the Armbian image also crashes with sudden USB errors in the logs.

krakenrf, 2022-08-26:
Maybe I spoke to soon. I now connected it to the bottom USB port on the DC barrel side, and it's been stable for 15 minutes. Before I was connected to the top port on the DC barrel side and it would always fail after 1-3 minutes. Will leave it on overnight and see what happens.

Could be sensitive to cable length too. I'm using a 1m cable. Maybe shorter cables will work better on the other USB ports.

xmutantson, 2022-08-26:
I believe this is the image I installed. specifically went with minimal because I didn't want the desktop overhead since this will be mounted in a car and controlled via SSH. 

https://wiki.odroid.com/odroid-n2/os_images/ubuntu/20210202

xmutantson, 2022-08-27:
I've got the kraken now, I hope I'm not experiencing the usb issues you describe but I cant come up with another explanation as to why it wont calibrate properly. How are you checking for the USB issues? which log files, keywords, etc should I be looking for?

krakenrf, 2022-08-27:
The symptom is that the UI updates will freeze, so it will continue to show uncalibrated.  From the krakensdr root folder you can do a `tail heimdall_daq_fw/Firmware/_logs/rtl_daq.log`, and you'll see numerous USB transfer issues logged indicating USB instability.

On my N2+, I've had it running Armbian https://www.armbian.com/odroid-n2/, plugged into the bottom left USB port overnight, and it's been stable over 12 hours.

So I think it's a combination of the official Ubuntu distro's that Odroid put out being buggy, plus the USB ports themselves not being stable, and only one port which is presumably the first port in the chain working properly with powered hub devices like the KrakenSDR.

krakenrf, 2022-08-28:
Running stable almost three days now, so I believe this specific combination is a solution for the N2.

Added a note to https://github.com/krakenrf/krakensdr_docs/wiki/11.-Appendix#tested-devices

krakenrf, 2022-09-01:
Stable about 6 days now. Will close this issue now as I think this is a solution. But if you're still having issues please reopen.

xmutantson, 2022-09-01:
Still running Ubuntu because reinstalling this OS would be less than ideal, I have noticed that it will periodically get "unlocked" and spend some time trying to calibrate. It looks like you describe the USB ports as being an issue. I don't see that issue in my logs when it does the instability thing. I've seen it on the kraken side taking anywhere from a few seconds (what i expect normal to be) to a few minutes at worst. The frame index resets when it does this. 

Is there some kind of debug switch I need to flip to see the USB issues in this file?

```
kameron@highwayman:~/krakensdr$ tail heimdall_daq_fw/Firmware/_logs/rtl_daq.log
Found Rafael Micro R820T/2 tuner
Found Rafael Micro R820T/2 tuner
Found Rafael Micro R820T/2 tuner
Found Rafael Micro R820T/2 tuner
Found Rafael Micro R820T/2 tuner
Allocating 12 (non-zero-copy) user-space buffers
Allocating 12 (non-zero-copy) user-space buffers
Allocating 12 (non-zero-copy) user-space buffers
Allocating 12 (non-zero-copy) user-space buffers
Allocating 12 (non-zero-copy) user-space buffers
```

krakenrf, 2022-09-01:
Is it running on your device normally apart from that 'unlocking'? I think the unlocking you're describing is just the periodic calibration check. Every few minutes it will briefly check for calibration, then go about it's normal business again.

There's no need to change debug settings to see USB transfer errors. They're serious enough to show up under any logging setting.

I believe in Odroids Ubuntu OS, if I connected to the 'good' USB port I wouldn't see transfer errors, but it would still be unstable.

xmutantson, 2022-09-01:
I did a short foxhunt and it did work. I did notice the app would sometimes show the red line stuck at 0 degrees but that did not persist or seem to affect the actual routing. I am connected to the 'good' usb port and haven't seen a transfer error in several hours.

krakenrf, 2022-09-01:
Ok great. Yeah it briefly being stuck at 0 degrees is most likely during the calibration check. Should be every 5 minutes, but you can make it less in the web GUI DAQ settings if you want.

xmutantson, 2022-09-01:
I think the 0 degrees was a lot longer than that but I was driving so was trying not to get information overload. I have a new bug for you:

Message from syslogd@highwayman at Aug 31 23:41:50 ...
 kernel:[ 8059.687864@1] NMI watchdog: BUG: soft lockup - CPU#1 stuck for 23s! [python3:3924]

3924 is python3 _daq_core/delay_sync.py

Still nothing about USB transfers in the rtl_daq.log file

xmutantson, 2022-09-01:
sorry for the fast followup, I'm now getting more lockup messages, and workqueue lockups <102 seconds. the latency reading from krakensdr's config page seems to match this at <102,000 ms. It's calibrating forever

krakenrf, 2022-09-02:
I didn't check the kernel messages, but yeah that's similar behavior with calibration failing that I was seeing on Odroid's Ubuntu. Unsure what they're doing different to other distros.

As you mentioned that you can't reinstall the OS, are you running a lot of other programs on the Odroid at the same time? Perhaps some other software is causing issues, or hogging CPU?

xmutantson, 2022-09-02:
No, the only process that was running and taking that much was the _daq_core/delay_sync.py at ~107% for the master process if i remember right. apart from that no other process went over 15% and the cores looked pretty ok. Load average was about 2.5-3 for the 5 minute, and on a hex core system that's fine. I didn't check kernel messages either, that popped up as a message in all connected terminals at the time.

The reason I don't want to reinstall the OS is because the system has already been configured with 3 wifi cards, policies to assign those correctly on start, netplan, and kismet for wardriving. If I have to I will but that means documenting everything I did to it since flashing it which is a bit of a pain.

krakenrf, 2022-09-02:
If delay_sync is at 107% that probably indicates its stuck on the correlation stage and can't calibrate because of the USB issue. I'll try to look more deeply into the problems with the Odroid distro later.

I also wonder if the need to use an external WiFi card could be causing issues with the Odroid. If the USB bandwidth is being shared with a high BW device like a WiFi card, it could become saturated, or at least increase the jitter enough that aligning USB samples for coherent becomes impossible.

With the Pi3 which shared WiFi bandwidth with the USB ports we used '[wondershaper](https://github.com/magnific0/wondershaper)' to limit WiFi bandwidth, to ensure there was enough bandwidth for the Kerberos (old model of Kraken).

krakenrf, 2022-09-02:
Reopening to remind myself to check the Odroid distro problems again.

xmutantson, 2022-09-02:
Out of the wifi cards, only one is active at a given time. There is one that connects to my phone's hotspot, or I will use the access point it creates. I will only ever connect one device to one or the other, my phone to run the kraken software. And one card in monitor mode searching for access points and logging them. 

If i was running high data transfer on all 3 cards I can see where that would be high bandwidth and mess with everything but I don't think that's the case.

krakenrf, 2022-10-06:
Ok retested the Odroid N2+ with the stock Odroid Ubuntu image and it is working perfectly well **IF** you use the bottom USB port next to the DC barrel jack, with a high quality cable. Any other USB port and it fails or only works temporarily.

I also tested with a USB WiFi stick installed and there are no problems with it serving the Kraken web GUI.

The brief 0 degree plots were just the auto calibration checks going on, which can last a few seconds every 5 minutes. 

I think the soft lockup bug isn't important. It just indicates that the Kraken software didn't allow access to other programs for a while, probably because it was hogging the system to ensure no samples were dropped during sample acquisition. That's normal and good as we want the Kraken software to run as a real time data acquisition system. We don't want other programs blocking the Kraken's ability to ingest data in real time.

So I'm closing this as resolved. I think in the end the only issue was that only one USB port on the Odroid N2 is capable of running the KrakenSDR. No idea why the others fail, I can only suspect the Odroid USB ports introduce some resistance or impedance which messes up USB comms for high data rate devices.

closed 2022-10-06 by krakenrf

krakenrf, 2022-10-07:
OK the N2 keeps surprising me with it's quirks. 

After ~1-2 hours of operation there is a stability problem on the N2 if a USB WiFi card is used. Samples will become misaligned at some point causing a minor sample misalignment. It's not a massive deal if you have periodic calibration turned on and checking at a reasonable timeframe, but for 5 or so minutes you could have bad results.

When run from the Ethernet port it's rock stable. 

I wonder if wondershaper could restrict the WiFi card enough to not cause a USB glitch...I will test that later.