krakenrf/krakensdr_doa #34: Android App won't connect to "KrakenSDR Server Settings"

royldean, 2022-09-24:
Where I'm at:
The DOA server starts, and I am able to access the settings website via the :8080 port on the host computer, which is an older HP laptop. It appears that I can make settings changes as well.

I am able to also access all 5 of the SDRs on the host computer using GQRX (while not running DOA server)

I am able to connect the DOA server host computer to the WIFI hotspot on my phone, and have a good connection (can websurf as expected).

I can open the KrakenSDR App on the hotspot phone, and for the most part, it seems to work. However, even after setting the server computer IP in the settings menu, I cannot access the KrakenSDR Server Settings. If I go to that portion of the App, I just get a white screen with the "return" arrow in the upper left.

royldean, 2022-09-24:
Ok, update.  I've tried on a second laptop (yet another old HP.... my work trashes these every few years so I have a bunch of them).   It works on this one.   Problem is this one is huge and there is no real way to power it in a car so it's of no use for direction finding.

What could prevent my other laptop from connecting?   Any suggestions?

krakenrf, 2022-09-25:
What is your phone and OS? If you just try browse KRAKEN_IP_ADDR:8080 with Chrome on the Android phone does it work? Might be some issue with the version of Chromium on your phone, or perhaps it's been locked down and cookies have been disabled? Cookies need to be enabled for the Dash library to work.

royldean, 2022-09-25:
Haven't tried the suggestions, yet, but some additional info:

OS on both computers running the KrakenSDR DOA server:  Ubuntu 20.04.
Phone is a Galaxy S10 5G, Android version 12.

Both the laptops and the phone are running Brave browsers.   I'll give the browser check a try later today.

royldean, 2022-09-25:
Ok, tried on both Brave and Chrome on the phone:

This site can't be reached.
The website http://192.168.48.156:8080/ might be temporarily down or it may have moved permanently to a new web address
ERR_CONNECTION_ABORTED

looks like there is something preventing this specific laptop from opening that port?

royldean, 2022-09-25:
Ok, update.   Upon further inspection, UFW was on.   I disabled it (to match the other laptop that IS connected to the app), but still no workie.

This time, when I try to open KrakenSDR Server Settings in the app, I get:
net::ERR_CONNECTION_REFUSED

Trying to open port 8080 on the phone browser, however, DOES connect to the settings!

krakenrf, 2022-09-25:
Ok nice, so it UFW was probably enabled from your work and it was probably blocking port 8080?

royldean, 2022-09-26:
> Ok nice, so UFW was probably enabled from your work and it was probably blocking port 8080?

*shrug*

Not sure.  I installed Ubuntu on both of these machines and have no memory or enabling and/or disabling UFW on either.  But one of them had it on, and it just so happed to be the one I wanted to use.

closed 2022-09-27 by krakenrf