krakenrf/krakensdr_doa #36: Automatic calibration of Kraken software for old Kerberos SDR

F4ASS, 2022-10-12:
Hello,
I looked for information on the automatic calibration carried out by RF coaxial switches and I fell back on the ckoval7 's fork which worked well with my old Kerberos version.
I don't know if this has been transposed for the Kraken software when using DQA kerberos example files?
I did not find any traces on the different supports and my GPIOs on pins 18 / 16 are not controlled during calibration.
Is there an implementation planned or did I miss something?

Sebastien

krakenrf, 2022-10-13:
Yes it should be supported out of the box by the new code. On the Pi4 image file, you just need to set the configuration file to the kerberos one, and it should automatically use the switches.

F4ASS, 2022-10-13:
I work with Pi 4 Image File V1.31 on a PI4 hardware.
There is no automtic switch command on P16 but only on P18.
it's sad not to have the **switch** and the **/switch** for command ?
Unless if the other command is elsewhere.
Thank you
Sebastien

F4ASS, 2022-10-13:
Hello. 
I found the problem
It's in rtl_daq.c file:
[image]

Change the line :       gpioWrite(24, 0);     by      gpioWrite(24, 1);

[image]

After do : make
[image]

For finish do reboot
And it's OK , you have Switch and /switch on pin 16 and 18 ;-)
Seb

krakenrf, 2022-10-14:
I don't fully understand this change. Are you connecting an external noise source through RF2 instead of using the internal noise source?

This change will switch in RF2 when the noise source is on.

F4ASS, 2022-10-15:
Hello,
No It's not an external noise source, but relays. In fact, the P16 output is always at 3,3v and this change to 0v when auto-cal. The P18 do nothing !
For my relay system I must use an 3.3v when i want auto-cal (Kreberos output to 50 Ohm).
For example you can see  the schematic here (but for my system I use real coaxial relays like Radiall for RF immunity and short M-M sma adaptators) :

[image]

Sebastien

krakenrf, 2022-11-01:
Closing as it appears resolved, and is a fairly unique custom hardware mod.

closed 2022-11-01 by krakenrf