krakenrf/krakensdr_doa #38: conda error

jhonnybonny, 2022-10-21:
after install on my pi4 i have conda error

_./kraken_doa_start.sh: line 4: conda: command not found
./kraken_doa_start.sh: line 5: conda: command not found_

_ubuntu@ubuntu:~/krakensdr_doa/heimdall_daq_fw/Firmware/_logs$ cat rtl_daq.log 
04:55:12 FATAL rtl_daq.c:494: The serial numbers of the devices are not yet configured, exiting.._

(but i do eeprom_init.sh )

also 

_ubuntu@ubuntu:~/krakensdr_doa/heimdall_daq_fw/Firmware/_logs$ cat delay_sync.log 
Traceback (most recent call last):
  File "/home/ubuntu/krakensdr_doa/heimdall_daq_fw/Firmware/_daq_core/delay_sync.py", line 43, in <module>
    import numba as nb
ModuleNotFoundError: No module named 'numba'_

krakenrf, 2022-10-22:
Did you install via the script, or some other way? Try running the script again if needed. For whatever reason it seems the conda install failed on your first try?

closed 2022-10-24 by jhonnybonny

krakenrf, 2022-10-24:
I see you've closed the issue. Are you able to let us know how it was resolved? Just in case this happens to anyone in the future?

jhonnybonny, 2022-10-27:
I just installed your PI original image )