krakenrf/krakensdr_doa #39: v1.322 Image, Antenna port mapping error? (low impact?)

n8rdf, 2022-10-30:
A possible mis-mapping of antenna ports between the CONFIG and SPECTRUM tabs may exist. To observe it
1. Set CONFIG tab to receive a known active signal and VFO Spectrum Calc to All Ch (TEST ONLY)
2. Connect an antenna to CH0 port ONLY (other ports not connected)
3. Observe SPECTRUM tab, active signal is shown correctly on CH0 (blue) trace only (OK)
4. On CONFIG tab change VFO Spectrum Calc to Single Ch
5. Observe SPECTRUM tab, CH0 (blue) trace shows no active signal (incorrect?)
6. Move antenna from CH0 to CH1. Observe SPECTRUM tab, active signal is shown as CH0 (blue) trace (incorrect?)
7. On CONFIG tab, change VFO Spectrum Calc to All Ch (TEST ONLY)
8. Observe SPECTRUM tab, active signal is shown as CH1 (orange) trace (correct?)

This issue is not likely to be seen if all antenna ports have antennas connected.  Please advise if my interpretation is incorrect.  Thanks.

krakenrf, 2022-10-30:
I think it's just because Single Ch mode actually takes data from CH1 and not CH0. The reason is that CH1 is a bit cleaner in terms of noise, so it's a cleaner way to view the spectrum. The CH0 noise has no real effect on the DFing ability.

I'll see if I can update the spectrum display to show that it's displaying Channel 1

krakenrf, 2022-11-05:
Fixed in c0d3f4d3bd4d6cda433e5a7bac1371d8f18a75fa

closed 2022-11-05 by krakenrf