krakenrf/krakensdr_doa #40: ZeroDivisionError: division by zero

Rich2020, 2022-10-31:
The WebGUI is crashing constantly and the terminal outputs this:

```
INFO:quart.serving:127.0.0.1:34926 GET / 1.1 200 652 8953
[2022-10-30 23:34:15,192] 127.0.0.1:34926 GET / 1.1 200 652 8953
INFO:quart.serving:127.0.0.1:36212 GET /_push 1.1 101 - 5994
[2022-10-30 23:34:16,182] 127.0.0.1:36212 GET /_push 1.1 101 - 5994
Traceback for <Task finished name='Task-160' coro=<Pusher.dispatch() done, defined at /home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/ctx.py:386> exception=ZeroDivisionError('float division by zero')> (most recent call last):
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/ctx.py", line 389, in wrapper
    return await websocket_context.app.ensure_async(func)(*args, **kwargs)
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/pusher.py", line 147, in dispatch
    await func(data['data'], client, data['id'])
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/dash.py", line 1261, in dispatch
    return
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/dash.py", line 1232, in call_callback
    return await loop.run_in_executor(None, runcoro, func(body, response, lock, client))  # %% callback invoked
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/concurrent/futures/thread.py", line 58, in run
    self.future.set_result(result)
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/_utils.py", line 252, in runcoro
    return e.value
  File "/home/rich/miniforge3/envs/kraken/lib/python3.9/site-packages/dash_devices/dash.py", line 1158, in add_context
    lock.release()
  File "/home/rich/krakensdr/krakensdr_pr/_UI/_web_interface/kraken_web_interface.py", line 1153, in display_page
    return [generate_config_page_layout(webInterface_inst), "header_active", "header_inactive", "header_inactive"]
  File "/home/rich/krakensdr/krakensdr_pr/_UI/_web_interface/kraken_web_interface.py", line 519, in generate_config_page_layout
    wavelength= 300 / webInterface_inst.daq_center_freq
ZeroDivisionError: float division by zero
```

krakenrf, 2022-10-31:
```ZeroDivisionError: float division by zero```

This usually indicates the DOA software was unable to grab data from the DAQ. Is the heimdall DAQ running?

krakenrf, 2023-02-10:
Closing due to no response, assuming it was a power issue.

closed 2023-02-10 by krakenrf