krakenrf/krakensdr_doa #41: Connection Status: Disconnected

ibootstrapper, 2022-11-01:
Hi,

We installed the kraken software via the script
```
wget https://raw.githubusercontent.com/krakenrf/krakensdr_docs/main/install_scripts/krakensdr_x86_install_doa_pr.sh
sudo chmod +x krakensdr_x86_install_doa_pr.sh
./krakensdr_x86_install_doa_pr.sh
```

first time we ran the software the kraken connected and the connection status was green. 

After trying the second time and following that the Connection Status shows Disconnected on the Web UI. 

We tried restarting the system and retry but that did not help.

krakenrf, 2022-11-01:
What device and setup do you have? Please be specific.

ibootstrapper, 2022-11-01:
Using the Kraken SDR device with a desktop and ubuntu

krakenrf, 2022-11-01:
How are you powering the KrakenSDR, and what are the specs of your power supply?

Can you check the logs at krakensdr_pr/krakensdr_pr/ui.log and heimdall_daq_fw/Firmware/_logs ?

krakenrf, 2022-11-01:
I think I found an issue with the x86 install script. The numba icc_rt optimized library that we used has an issue and it causes JIT numba compilation issues on some machines. The fix is to remove the icc_rt optimization and run the software again.

Can you try 

```conda activate kraken
conda remove icc_rt
```

Select Y when it asks.

Then go through all the folders in heimdall and krakensdr_pr and delete the __pycache__ folders.  There should be __pycache__ folders to remove in the Firmware, _daq_core folders in heimdall, and the _receiver, _signal_processing, and _UI/_web_interface folders in krakensdr_pr.

After that try running again, and the numba compilation should succeed.

ibootstrapper, 2022-11-05:
thanks, was a faulty power supply .

closed 2022-11-05 by ibootstrapper