krakenrf/krakensdr_doa #42: V1.322  image, Krakensdr Map function stopped working? (2 receivers)

n8rdf, 2022-11-04:
I have 2 Krakensdr, each on it's own Rpi4. Both work fine with the Android app. Both have briefly worked on the Krakensdr Map site (but usually one or the other, not both) in the last few days. To try to debug the intermittent behavior today I reflashed a fresh V1.322 image, edited wpa_supplicant.conf for my wifi hotspot and confirmed good operation with the Android app when configured for DoA Data Format= Kranken app or = Kraken Remote Pro (with api keys).  When configured for Remote Pro and logged into the Map site, Start Remote data does not see either receiver.  Screen shots of my Configs are attached.  Any advice is appreciated.

[image]
[image]

krakenrf, 2022-11-05:
Did you remember to restart after saving the config with the API key there?

n8rdf, 2022-11-05:
Oops, probably not, but eventually power cycled during troubleshooting. I powered up both today, N8RDF5 Mapped fine first try, N8RDF6 never did. I paused processing on 5, stopped and restarted 6 and the Map data multiple times to try to get it mapping with no luck. I restarted processing on 5 and Map data cannot get it to map again.  I'm migrating to the v1.4 image and will start again from scratch.

n8rdf, 2022-11-05:
Upgraded N8RDF5 to image V1.4. Both 5 and 6 now mapping correctly. At least until 5 processing is Stopped, then Started. This seems to "break" the Map display (both bearings disappear, replaced with pink blinking dot until 5 is re-Started and Map Data is Deleted and Recalculated). It seems the second receive station data is not visible if the first receive station data is not available. Only the first receiver is ever heat mapped, so no heat map when it's not available.

n8rdf, 2022-11-05:
Upgraded N8RDF6 to image v1.4, no longer mapping. Reverted 6 to v1.322, still not working. Multiple reboots, system power cycles, Map site logoff-login, Data stop/start/delete. no success. Rpi DOA screen shows valid, updating DOA estimation in all cases. No issues seen with Android app. No issues seen with wifi connection, speed.

n8rdf, 2022-11-05:
Both stations work fine when configured for the RDFMapper site

krakenrf, 2022-11-07:
Did you happen to set the coordinates from 0,0 to somewhere else after you started the system with the API key? There seems to be a bug that causes a huge lag if that happens. Looking into fixing it soon.

n8rdf, 2022-11-07:
Not to 0,0,0 but I sometimes change the static heading or the lat/lon during mapping. It seemed to update the map fine (for the station that was uploading successfully).  I was also playing with GPS on the Rpi usb but it seemed it created more problems with the mapping.

n8rdf, 2022-11-19:
Both stations mapping fine today.

krakenrf, 2022-11-21:
Excellent. I think it was more to do with changes pushed to the web system though, nothing in the image file itself.

closed 2022-11-21 by krakenrf