krakenrf/krakensdr_doa #43: ZeroDivisionError: division by zero

golledgekb, 2022-11-05:
running into this error 

bash kraken_doa_start.sh 
Shut down DAQ chain ..
Config file check bypassed [ WARNING ]
kernel.sched_rt_runtime_us = -1
Desig FIR filter with the following parameters: 
Decimation ratio: 1
Bandwidth: 1.00
Tap size: 1
Window function: hann
FIR filter ready
Transfer funcfion is exported to :  _logs/Decimator_filter_transfer.html
Coefficients are exported to:  _data_control/fir_coeffs.txt
Starting DAQ Subsystem
Output data interface: Shared memory
      )  (     
      (   ) )  
       ) ( (   
     _______)_ 
  .-'---------|
 (  |/\/\/\/\/|
  '-./\/\/\/\/|
    '_________'
     '-------' 
               
Have a coffee watch radar
Starting KrakenSDR Direction Finder
Web Interface Running at 0.0.0.0:8080
PHP Data Out Server Running at 0.0.0.0:8081
pi@pi:~/krakensdr_doa $ gpsd Available
 * Serving Quart app 'kraken_web_interface'
 * Environment: production
 * Please use an ASGI server (e.g. Hypercorn) directly in production
 * Debug mode: False
 * Running on http://0.0.0.0:8080 (CTRL + C to quit)
[2022-11-05 03:29:21,127] Running on http://0.0.0.0:8080 (CTRL + C to quit)
[2022-11-05 03:29:28,623] 192.168.193.11:62481 GET / 1.1 200 661 20246
[2022-11-05 03:29:28,664] 192.168.193.11:62481 GET /assets/style.css 1.1 200 1309 11230
[2022-11-05 03:29:28,705] 192.168.193.11:62482 GET /_dash-component-suites/dash_bootstrap_components/_components/dash_bootstrap_components.v1_1_0m1667614394.min.js 1.1 200 59350 33785
[2022-11-05 03:29:28,767] 192.168.193.11:62483 GET /_push 1.1 101 - 4618
[2022-11-05 03:29:28,781] 192.168.193.11:62482 GET /assets/favicon.ico 1.1 200 27903 26828
[2022-11-05 03:29:29,295] 192.168.193.11:62482 GET /assets/kraken_interface_bw.png 1.1 200 10001 15534
Traceback for <Task finished name='Task-62' coro=<Pusher.dispatch() done, defined at /home/pi/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/ctx.py:386> exception=ZeroDivisionError('division by zero')> (most recent call last):
ZeroDivisionError: division by zero
[2022-11-05 03:29:29,821] 192.168.193.11:62481 GET /config 1.1 200 661 10963
[2022-11-05 03:29:29,950] 192.168.193.11:62484 GET /_push 1.1 101 - 6421
Traceback for <Task finished name='Task-96' coro=<Pusher.dispatch() done, defined at /home/pi/miniforge3/envs/kraken/lib/python3.9/site-packages/quart/ctx.py:386> exception=ZeroDivisionError('division by zero')> (most recent call last):
ZeroDivisionError: division by zero

HB9IBI, 2022-12-06:
Hello [golledgekb]
I had this kind of problem on many PC i've tried. Now i run a Intel Nuc I3, without issue.
Probably an hardware/driver incompatibility ...
Good luck !
François

krakenrf, 2022-12-07:
What hardware, power supply and USB cable quality/length? This error simply indicates that the DSP code received no data, and the most usual cause of this is because the tuners did not all start up due to some problem with the power supply being insufficient, or poor quality cables.

closed 2022-12-11 by golledgekb