krakenrf/krakensdr_doa #44: Map server feature request: Multiple receivers, Indicate which bearing is which receiver

n8rdf, 2022-11-05:
When using multiple receivers, it's not clear which map bearing is for which receiver. Especially important for mobile Dfing. Suggest adding a bearing source label or some method to correlate receivers with map bearings display.

Tysonpower, 2022-11-07:
are you talking about the bearing lines or what exactly do you want a label for? Maybe a screenshot would be helpful :)

n8rdf, 2022-11-07:
[image]
You have a list of receivers and multiple bearings displayed but can't see which bearing is from which receiver unless you click on one. That might be ok for fixed stations but for mobile(s) it would be nice to have the station name on the map somewhere near the bearing line.

Tysonpower, 2022-11-07:
Thx for the clearification, added to the TODO list :)

n8rdf, 2022-11-30:
Datapoint Info popup could/should have receiver ID. (Also, the red hexagons are so small I didn't know they existed until today since I never zoomed in close enough to see them.)
[image]

n8rdf, 2023-04-16:
Any update on this? I'm trying to put both krakensdr on line and the mapper is confusing to read. Thanks.
[image]

Tysonpower, 2023-07-23:
working on it

Tysonpower, 2023-07-25:
The stations should be marked with a marker and Name on the Map now with the latest update.

closed 2023-07-25 by Tysonpower