krakenrf/krakensdr_doa #46: Unsupported Compression Method: 14

cisrgistics, 2022-11-06:
About to do a road trip. Had some ULA signals lined up. Noticed the update. Downloaded. BalenaEtcher said the v1.4 zip file was unsupported. Realizing my age and level of stupidity I thought maybe I touched something wrong. Tried directly from the link to the file on github. No dice. BalenaEtcher cried again. I can't update. Using a Win10 machine with MS Edge browser....Also, why is it the other version zip files have disappeared? CYA decisions or we're not allowed to blend code between versions?

n8rdf, 2022-11-06:
My Balena worked fine with v1.4 image.

cisrgistics, 2022-11-06:
Since your zip didnt have any issues, how about you upload your copy to my folder? https://1drv.ms/u/s!AoICwvibJ8iOhMNYOUW31ZKw9_vJGQ?e=MTJvza
[image]

cisrgistics, 2022-11-06:
and this
[image]

cisrgistics, 2022-11-06:
and this effort
[image]

cisrgistics, 2022-11-06:
Which has the same result:
[image]

n8rdf, 2022-11-06:
Did you try downloading the image zip first, then extracting from the local copy? That's how I did it.

krakenrf, 2022-11-07:
Interesting looks like whatever compression library Etcher is using doesn't support LZMA compression. We switched to that for greater compression efficiency because otherwise it was going over the 2GB limit for GitHub. Will look at changing the compression method, or uploading elsewhere.

For now please just unzip the file first with 7zip.

We had to remove the older images due to a very unfortunate legal regulation requirement for passive radar which we just became aware of. We need to wait on the lawyers to sort that out before we can say or do anything else.

krakenrf, 2022-11-07:
Ok updated to .xz LMZA2 compression which etcher seems to support. Should be OK now.

closed 2022-11-07 by krakenrf