krakenrf/krakensdr_doa #49: DOA Issue: Kraken stops functioning set at 1.68 ghz (NOAA wx balloons)

n8rdf, 2022-11-08:
V1.4 image, DOA is working fine at 853.85mhz (antenna array supports 700-1700 mhz). If I change frequency to 1.680 ghz, the config shows errors (see photo) and won't recover without Software Restart.
[image]

krakenrf, 2022-11-09:
Interesting, I was able to replicate it. This needs more investigation, but for now I was able to get around the issue by tuning to 1500 MHz first, and then to 1680 MHz.

Could be something to do with VCO current needing to ramp up faster with larger frequency changes.

n8rdf, 2022-11-09:
Wow, that worked here too. Maybe I can chase some balloons now 😃

darksidelemm, 2022-11-11:
(Sorry to hijack the issue!)
@n8rdf What area of the US are you still in that is still flying 1680 MHz sondes? Most have switched to 400 MHz now.
You also do know that you can just decode the telemetry from these :-)

darksidelemm, 2022-11-11:
If you are located in Michigan as your QRZCQ profile suggests, then both launch sites here are now running on the 400-406 MHz band:
White Lake, MI - 404 MHz, Vaisala RS41
Gaylord, MI - 403.01 MHz, Graw DRM-17

n8rdf, 2022-11-11:
Thanks for the info, that explains the epic fail first try 😃. Guess I watched the wrong youtube videos.

darksidelemm, 2022-11-11:
The switch to 400 MHz sondes has really only happened in the last 12 months, so whatever you found on youtube is probably out of date :-)