krakenrf/krakensdr_doa #53: DOA Issue: GPS location "Disabled" at startup (Image v1.4)

n8rdf, 2022-11-30:
I have the Station Information, Location Source: GPS in my saved config file. At bootup, the system starts operating OK but with the GPS = Disabled status.  I have edited /etc/default/gpsd as 
# Default settings for gpsd.
START_DAEMON="true"
GPSD_OPTIONS="-G"
DEVICES="/dev/gps0"
USBAUTO="true"
GPSD_SOCKET="/var/run/gpsd.sock"
and confirmed the gpsd Daemon is running at startup. 
However, DOA appears to be using the last (possibly invalid) GPS location for DOA reports, both locally and on the Kraken RF map site.
 Deseleting and reselecting GPS as the Location Source appears to allow the GPS to Connect. However, I'm trying to configure a remote DF station that can operate from bootup without user intervention.

krakenrf, 2022-12-01:
What model GPS puck are you using?

n8rdf, 2022-12-01:
[image]
Ublox 7 and g-mouse ipx6

krakenrf, 2022-12-02:
Ah is this on the Pi4 image, or a custom setup?

n8rdf, 2022-12-02:
Pi4 image 1.4

krakenrf, 2023-02-02:
Fixed in https://github.com/krakenrf/krakensdr_doa/commit/f1ac9e469be9b7df8c09ab20914952bfc497ec80, will be updating the Pi image within the next few days.

closed 2023-02-02 by krakenrf