krakenrf/krakensdr_doa #60: DOA Image v1.43 install or path error?

n8rdf, 2023-02-11:
DOA won't start at bootup per photo. Possible path error (or operator error, total linux idiot 😃)?

[image]

krakenrf, 2023-02-12:
Is this the stock Pi4 image? On the Pi4 image the root kraken dir is krakensdr_doa. You can re-run the start script from in there.

Don't see why the stock image wouldn't just start though. Tested on two Pi4's without issue. Does V1.41 run OK?

n8rdf, 2023-02-12:
Yes, stock image and v1.41 worked fine. Again, I don’t speak linux, but why is” krakenrf” in script path but not seen in ls command in photo?

n8rdf, 2023-02-12:
Maybe because I gave it a unique node name in the imager? I have two krakens on my (zerotier) network.

krakenrf, 2023-02-13:
Changing the hostname shouldn't make a difference. But it sounds like you might be using the Raspberry Pi imager? I have heard of that corrupting the image. Can you try burn it with a more standard imager like Balena Etcher, or Win32 Disk Imager, and then change the hostname after?

closed 2023-03-21 by krakenrf