krakenrf/krakensdr_doa #66: ui.log error

galaris, 2023-03-28:
Got this in my doa root folder, in ui.log:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/krakenrf/miniforge3/envs/kraken/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/home/krakenrf/krakensdr_doa/krakensdr_doa/_signal_processing/krakenSDR_signal_processor.py", line 236, in run
    single_ch = self.processed_signal[1, :]
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed
INFO:quart.serving:Running on http://0.0.0.0:8080 (CTRL + C to quit)

I am having some problems with the software, however I dont know if its related to this error message at all or not.

krakenrf, 2023-03-31:
Seems like the Kraken disconnected as the data vector was empty. Could it simply be the power or USB being bumped? Loose cables? Or a failing power supply?

godsic, 2023-04-10:
@galaris I hope #72 fixes this by skipping empty data frames and dumping a clear message to the log.

closed 2023-06-12 by krakenrf