krakenrf/krakensdr_doa #92: Cannot get DOA to work

thanosradio, 2023-07-25:
I am very new to this and had no prior experience with using a Raspberry Pi. I got a Kraken, Pi 4, and flashed the latest DOA script onto a microSD. I followed the steps on the krakensdr_doa page, and when I get to the web interface on a phone, it constantly shows that it's not connected or restarting. The spectrum and DOA estimation tabs don't show anything either. I'm just not really sure what to do at this point; any help would be much appreciated.

*I am using the Pi 4 Wi-Fi Hotspot option

krakenrf, 2023-07-25:
Is the KrakenSDR plugged in to the Pi 4 and powered?

thanosradio, 2023-07-26:
Yes, I've tried a wall outlet plugin as well as an external battery that was the recommended 5W 2.4v USB-C type.

krakenrf, 2023-08-09:
Please try the tuner test and the power supply test on the KrakenSDR troubleshooting Wiki, and letme know the results https://github.com/krakenrf/krakensdr_docs/wiki/07.-KrakenSDR-Troubleshooting#tuner-test

krakenrf, 2023-09-05:
Any updates, or can I close this?

thanosradio, 2023-09-05:
Hi,

Apologies for the delay. I'm away from home on a work trip and I'll be
unable to work on troubleshooting the Kraken until October.

closed 2024-02-26 by krakenrf