krakenrf/krakensdr_doa #101: Add ability to download log files from UI

krakenrf, 2023-09-07:
Note for future work. Will be useful for customer debugging.

godsic, 2023-09-15:
@krakenrf Shell we just put them in `_share`?

krakenrf, 2023-09-25:
Maybe just a symlink to them will work in _share?

godsic, 2023-09-25:
@krakenrf Links to shared folders are considered to be bad security practices. Maybe we can create a log sub folder inside _share and write there directly?

krakenrf, 2023-09-26:
Hm but we also need to be able to get the logs out from the heimdall Firmware/_logs folder, and I don't think it's a good idea to make heimdall write it's logs to the krakensdr_doa/_share folder, since heimdall can be run standalone without krakensdr_doa. 

Perhaps there needs to be a script that periodically copies the logs over. But security wise, what's the difference between a symlink to the logs in another folder, vs copying/writing them directly?

krakenrf, 2023-09-26:
Also just noting if we're thinking about security, the way Local Data Recording is downloaded will need looking at too, since it's possible to traverse the directory structure via the filename and download anything. Probably just got to sanitize ../

PR #103 (2023-11-06): Share heimdall_daq_fw and krakensdr_doa logs
Compared to ssh, it is a bit more practical for end users to access logs via http. So lets move them to "_share/logs", including periodically syncing heimdall logs there. If legacy php-based web server is used, then user would need to access particular log via explicit path, e.g., http://localhost:8081/logs/krakensdr_doa/ui.log , while miniserver-based server would present a simplistic gui where logs can easily be navigated to. Fixes #101
closed 2023-11-06 by krakenrf