krakenrf/krakensdr_doa #131: Manuall install fail at cargo install miniserve

RasmusEk, 2024-03-15:
runing on Raspberrie pi 4 with Raspberry Pi OS Lite x64 I try to do a manuall install

after updating i install cargo and rustc through
sudo apt -y install nodejs jq rustc cargo

running 
sudo cargo install miniserve i get error due to rustc beeing v1.63.0 needing rustc 1.65 or newer try running cargo with -locked

runing
cargo install --locked miniserve

I get following error message:
warning: package 'iana-time-zoe v0.1.59' in cargo.lock is yanked in registry 'crates-io', consider running without --locked
error: failed to compile 'miniserve v0.26.0', intermediate artifacts can be found at '/tmp/cargo-installibm5Q2

caused by:
package 'zstd-sys v2.0.9+zstd.1.5.5' cannot be built becouse it requires rustc 1.64 or newer, while the currently active rustc version is 1.63.0

my rustc version is 1.63.0
my cargo version is 1.65.0

krakenrf, 2024-03-18:
I'll look into why this is suddenly happening, but the install of cargo and miniserve can actually be ignored. It's an optional extra for those with third party remote control setups.

RasmusEk, 2024-03-18:
Thank you, I dont need these options and will try procceding without miniserver when I do another manual install

R0hero, 2024-08-16:
I had this same issue while installing it manually.

It seems that the rustc version install using 
`sudo apt install rustc`
is outdated for the rest of the required packages in miniserve.
This can be fixed if Rust is installed directly from Rust.
`curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`

This will install the newest version of Rust. 

Note that this may break other dependencies of Rust in the Raspbian OS if something is developed using the Rust version accessible for Raspbian OS.