krakenrf/krakensdr_doa #132: Python3 issue ModuleNotFoundError: No module named 'numpy'

RasmusEk, 2024-03-15:
I'm running the image version "krakensdr_pi4_01_March_2024.img.xz" on the raspberry pi 4 as Doa and DSP

I am reciving a message called:

Config file check bypassed [ WARNING ]

as I am booting up

cd ~/krakensdr_doa/
./kraken_doa_start.sh

as far as I know the error comes from

~/krakensdr/heimdall_daq_fs/Firmware/ini_checker.py

since 

../Firmware/daq_synthetic_start.sh

crashes and reads out that it crashed on line 6 at ini_checker.py which reads:

import numpy as np

trying to run the same import in python interpretator gives me:

Traceback (most recent call last):
      File "<stdinn>", line 1, in <module>
ModuleNotFoundError: no module named 'numpy'

I have reinstalled numpy through

conda install numpy

gotten the same error made sure there was a $PATH to usr/bin aswell as added a path to usr/bin/python

trying

which numpy 

returns no result

I have not worked in python before but I belive that there is a pathing issue or as suggested in "https://forum.krakenrf.com/t/doa-start-script-multiple-errors/166/8" there might be a issue with conda (I'm not using a manual install as in the thread)

how can I continue my trubleshooting?

RasmusEk, 2024-03-18:
When you are talking about the script are you refering to

./kraken_doa_start.sh

that does not run on boot I seem to have issues with kraken not optaining port 8080 kraken requiering multiple reboots and other conectivity issues. Everytime I start 

./kraken_doa_start.sh

I get the error message Config file check bypassed [ WARNING ]

I try to find why I get that readout since a fresh install does not remove it and I get the same errorcode on my secound Raspberry PI 4

krakenrf, 2024-03-18:
> I get the error message Config file check bypassed [ WARNING ]

This is normal, in the current versions the config check is disabled.

Can you look at the log files in the _share folder, and in heimdall in the Firmware/_logs folder? That will tell you more information.

But 99% of the time when someone is having trouble booting the software from the pre-made image it's because either the USB data cable is poor quality, or because the power supply is not strong enough for the Kraken.

RasmusEk, 2024-03-23:
You can close this thread there seem to be nothing wrong and I was just confusing myself

closed 2024-03-23 by RasmusEk