krakenrf/krakensdr_doa #138: Possible issue in output format for DF Aggregator

alphafox02, 2024-04-12:
I’ve not tried myself yet, but based on commits to this repo in terms of output format changes, is it possible to check that the DFA output is still correct? 

I will check myself soon, but noticed this ticket and thought I’d make mention here.

https://github.com/ckoval7/df-aggregator/issues/12

krakenrf, 2024-04-27:
Forgot to reply here. The DF-Aggregator output is now output as doa.xml.

closed 2024-04-27 by krakenrf

alphafox02, 2024-04-27:
I may be missing something, I just pulled latest Kraken git code for testing. I can see everything is running and localhost:8081/DOA_value.html appears to be offering up results, but DF-Aggregator can’t connect to it like it used to. Is there a different url to you’re saying or does changes in DF Aggregator’s code need to happen?

alphafox02, 2024-04-27:
I see now. Changed to DFA output and found on localhost:8081/doa.xml and DFA connection works.