krakenrf/krakensdr_doa #155: Fresh install unsing script on Rpi5 inop.

HookdomPonix, 2025-02-15:
I'm running the latest Rpi5 factory 64 bit image and i'm not getting a working install.  

The system never starts listening on port 8080.   I do get listening on port 8081.  

Nothing is listening on 8080 already either.  

Steps:  

Install Raspbian 64 bit latest on RPI5.
Create krakensdr user.
Set password.

Run this script set:  

`wget https://raw.githubusercontent.com/krakenrf/krakensdr_docs/main/install_scripts/krakensdr_aarch64_install_doa.sh
sudo chmod +x krakensdr_aarch64_install_doa.sh
./krakensdr_aarch64_install_doa.sh`

Then cd to the kraken_doa directory and run the start script.  

I can see my device in lsusb just fine.  

Heimdall logs. show the system calibrating itself for example, so I think the Heimdall side is working.  

Thoughts?

HookdomPonix, 2025-02-15:
I modified the container image to build on the RPI5/CM5 as well.   Still getting the same error inside the container:  

`root@type4-1:/var/lib/containers/storage/overlay/00cc5831efaaf95cefb14f55d13d065bce62dbf7caa5ad088eda6f6e09ca488d/merged/root/krakensdr/krakensdr_doa/_share/logs/krakensdr_doa# cat ui.log 
gpsd Available
Traceback (most recent call last):
  File "/root/krakensdr/krakensdr_doa/_ui/_web_interface/app.py", line 22, in <module>
    from maindash import app
  File "/root/krakensdr/krakensdr_doa/_ui/_web_interface/maindash.py", line 27, in <module>
    waterfall_fig = init_waterfall(web_interface)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/krakensdr/krakensdr_doa/_ui/_web_interface/waterfall.py", line 13, in init_waterfall
    go.Heatmapgl(
    ^^^^^^^^^^^^
  File "/root/miniconda/envs/kraken/lib/python3.11/site-packages/plotly/graph_objects/__init__.py", line 303, in __getattr__
    return orig_getattr(import_name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/miniconda/envs/kraken/lib/python3.11/site-packages/_plotly_utils/importers.py", line 39, in __getattr__
    raise AttributeError(
AttributeError: module 'plotly.graph_objects' has no attribute 'Heatmapgl'. Did you mean: 'Heatmap'?
root@type4-1:/var/lib/containers/storage/overlay/00cc5831efaaf95cefb14f55d13d065bce62dbf7caa5ad088eda6f6e09ca488d/merged/root/krakensdr/krakensdr_doa/_share/logs/krakensdr_doa# `

HookdomPonix, 2025-02-15:
Okay - so this is because a function was removed from plotly in the latest versions.   All new installations should probably break under this model.   This script has been modified to revert to an earlier version of plotly and now function resumes.  

`#!/bin/bash

# Prevent sudo timeout
sudo -v # ask for sudo password up-front
while true; do
  # Update user's timestamp without running a command
  sudo -nv; sleep 1m
  # Exit when the parent process is not running any more. In fact this loop
  # would be killed anyway after being an orphan(when the parent process
  # exits). But this ensures that and probably exit sooner.
  kill -0 $$ 2>/dev/null || exit
done &

sudo apt update
sudo apt -y install build-essential git cmake libusb-1.0-0-dev lsof libzmq3-dev clang php-cli nodejs gpsd libfftw3-bin libfftw3-dev

cd
git clone https://github.com/krakenrf/librtlsdr
cd librtlsdr
sudo cp rtl-sdr.rules /etc/udev/rules.d/rtl-sdr.rules
mkdir build
cd build
cmake ../ -DINSTALL_UDEV_RULES=ON
make
sudo ln -s ~/librtlsdr/build/src/rtl_test /usr/local/bin/kraken_test

echo 'blacklist dvb_usb_rtl28xxu' | sudo tee --append /etc/modprobe.d/blacklist-dvb_usb_rtl28xxu.conf

cd
git clone https://github.com/krakenrf/Ne10
cd Ne10
mkdir build
cd build
cmake -DNE10_LINUX_TARGET_ARCH=aarch64 -DGNULINUX_PLATFORM=ON -DCMAKE_C_FLAGS="-mcpu=native -Ofast -funsafe-math-optimizations" ..
make

cd
wget https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-Linux-aarch64.sh
chmod ug+x Miniforge3-Linux-aarch64.sh
./Miniforge3-Linux-aarch64.sh -b

export PATH=~/miniforge3/bin/:$PATH
eval "$(conda shell.bash hook)"
conda init bash
conda config --set auto_activate_base false

conda create -y -n kraken python=3.9.7
conda activate kraken

conda install -y scipy==1.9.3
conda install -y numba==0.56.4
conda install -y configparser
conda install -y pyzmq
conda install -y scikit-rf
conda install -y scikit-image
conda install -y gitpython
conda install -y plotly==5.23.0
cd
mkdir krakensdr_doa
cd krakensdr_doa

git clone https://github.com/krakenrf/heimdall_daq_fw
cd heimdall_daq_fw

cd ~/krakensdr_doa/heimdall_daq_fw/Firmware/_daq_core/
cp ~/Ne10/build/modules/libNE10.a .
cp ~/librtlsdr/build/src/librtlsdr.a .
cp ~/librtlsdr/include/rtl-sdr.h .
cp ~/librtlsdr/include/rtl-sdr_export.h .
make

conda install -y pandas
conda install -y orjson
conda install -y matplotlib
conda install -y requests

pip3 install dash_bootstrap_components==1.1.0
pip3 install quart_compress==0.2.1
pip3 install quart==0.17.0
pip3 install dash_devices==0.1.3
pip3 install pyargus
pip3 install gpsd-py3
pip3 install plotly==5.23.0
conda install -y dash==1.20.0
conda install -y werkzeug==2.0.2

cd
cd ~/krakensdr_doa
git clone https://github.com/krakenrf/krakensdr_doa
cp krakensdr_doa/util/kraken_doa_start.sh .
cp krakensdr_doa/util/kraken_doa_stop.sh .

# OPTIONAL: Install Kraken TAK Server
git clone https://github.com/canaryradio/Kraken-to-TAK-Python`

krakenrf, 2025-02-18:
Thanks for catching that depreciation. I've pinned plotly in the install scripts now.

closed 2025-02-18 by krakenrf