krakenrf/krakensdr_doa #157: Port 8080 doesn't open

CurtisDufour, 2025-03-28:
I just followed the github instructions to do the automated install with the krakensdr_x86_install_doa.sh file on a fresh install of DragonOS. I don't see any issues with dependencies or errors through the process, and then when I start the kraken_doa_start.sh, it also does not give me any errors. However, port 8080 does not seem to be opening. 

[image]

When I run nmap on my localhost, I see these ports open:

[image]

and when I navigate to port 8081 to see what's going on, it gives me this error:

[image]

Tysonpower, 2025-03-28:
For the normal ui the correct port is 8080, not 8081 - pls try 8080, in your screenshot it says 8081 :)

CurtisDufour, 2025-03-28:
Yes, I understand that. But it isn't open, as you can see in the nmap results. There is nothing when I go to port 8080.

RadioAndrea, 2025-03-29:
Can confirm, same issue after installing everything manually, building with a working version of KFS.

```bash
[user@hostname kraken]$ nmap -p 8081,8080 127.0.0.1
Starting Nmap 7.92 ( https://nmap.org ) at 2025-03-28 23:17 EDT
Nmap scan report for localhost (127.0.0.1)
Host is up (0.000090s latency).

PORT     STATE  SERVICE
8080/tcp closed http-proxy
8081/tcp open   blackice-icecap

Nmap done: 1 IP address (1 host up) scanned in 0.02 seconds
```

I am investigating, but the python dependencies are in really bad shape right now.

Tysonpower, 2025-03-29:
I installed the software on a PI 400 with current Pi OS lite using the installation script without any issues yesterday, i guess this is an issue with the non script based manual installation as you mentioned. Why not use the install script?

I guess some more info about the exakt instructions you used (link maybe) as well as the hardware would be helpful.

RadioAndrea, 2025-04-01:
> I installed the software on a PI 400 with current Pi OS lite using the installation script without any issues yesterday, i guess this is an issue with the non script based manual installation as you mentioned. Why not use the install script?

Because I am not running on an RPI. I also tried to use the container build, however this is broken due to #158 and either the container build needs to be fixed to work around it or the KFR updated.

I switched to manual installation since I can build each portion individually rather than mucking around with the container build script.

> I guess some more info about the exakt instructions you used (link maybe)

The install instructions are right here, in the repo

https://github.com/krakenrf/krakensdr_doa?tab=readme-ov-file#manual-install

and

https://github.com/krakenrf/heimdall_daq_fw#manual-step-by-step-install

and

https://github.com/krakenrf/kfr#linux-macos-other

except as you can see in #158, the kraken KFR port is broken with the latest clang-19 so you MUST either use an older clang or update KFR from upstream and manually enable capi builds using the new KFR build targets.

> as well as the hardware would be helpful.

x86 AMD

RadioAndrea, 2025-04-01:
@CurtisDufour 

I managed to get the webUI started, and accessible on port 8080.

I am using way more up-to-date dependencies than the install instructions, and so you will have to do some minor code changes to account for this.

I really want this to run on x86, so I'm going to keep plugging at it, if I can fix, will open a PR.

For now, here's the python deps I'm using (not in conda, using UV manager in pycharm as I work to trace down the issues, copied from my pyproject.toml)

```toml
[project]
name = "krakensdrdoa"
version = "1"
dependencies = [
    "configparser==7.2.0",
    "dash==3.0.1",
    "dash-bootstrap-components==2.0.0",
    "dash-core-components==2.0.0",
    "dash-devices==0.1.3",
    "dash-html-components==2.0.0",
    "dash-renderer==1.9.1",
    "future==1.0.0",
    "gpsd-py3==0.3.0",
    "matplotlib==3.10.1",
    "numba==0.61.1rc1",
    "orjson==3.10.16",
    "pandas>=2.2.3",
    "plotly==6.1.0b0",
    "pyargus==1.1.post1",
    "pyzmq==26.3.0",
    "quart-compress==0.2.1",
    "requests==2.32.3",
    "scikit-rf==1.6.2",
    "scipy==1.15.2",
    "werkzeug==2.0.2",
]

```

CSSAAmufon, 2025-06-14:
Is there someone who can help solve my problem?  Unable to get the system running....

Shut down DAQ chain ..
Config file check bypassed [ WARNING ]
kernel.sched_rt_runtime_us = -1
Desig FIR filter with the following parameters: 
Decimation ratio: 1
Bandwidth: 1.00
Tap size: 1
Window function: hann
FIR filter ready
Transfer funcfion is exported to :  _logs/Decimator_filter_transfer.html
Coefficients are exported to:  _data_control/fir_coeffs.txt
Starting DAQ Subsystem
Output data interface: Shared memory
      )  (     
      (   ) )  
       ) ( (   
     _______)_ 
  .-'---------|
 (  |/\/\/\/\/|
  '-./\/\/\/\/|
    '_________'
     '-------' 
               
Have a coffee watch radar

Remote Control is DISABLED
To enable Remote Control please install miniserve and jq.
Then change 'en_remote_control' setting in _share/settings.json file to 'true'.
Finally, apply settings by restarting the software.

Starting KrakenSDR Direction Finder
Web Interface Running at 0.0.0.0:8080
Data Out Server Running at 0.0.0.0:8081
TAK Server NOT Installed

I need to get this running very soon for a skywatch trip.
Dave